scottdarch/Arturo #5: Arduino 1.6 moves hardware/boards.txt

neilmb, 2015-04-01:
The 1.5/1.6 versions of the Arduino IDE have modularized the entire structure of different cores, including the location of boards.txt. On Mac OS X 10.10, ano doesn't find any boards.txt:

    nmb@garble ~/tmp/blink % ano list-models
    Searching for Board description file (boards.txt) ... FAILED
    Board description file (boards.txt) not found. Searched in following places:
      - /Applications/Arduino.app/Contents/Resources/Java/hardware/**
      - /usr/local/share/arduino/hardware/**
      - /usr/share/arduino/hardware/**

The entire boards.txt infrastructure is now in subdirectories of $HOME/Library/Arduino15/packages/arduino. See https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification for more details.

scottdarch, 2015-04-01:
We had this fixed when we did the branch (It was the reason for the branch). Did you try doing a clean first? If you had built using ino you'd be using the incorrect makefiles.

scottdarch, 2015-04-01:
I just merged a version of ano that uses an entirely new build directory name ".build_ano". It is version 0.4.4. Please try this and confirm the problem is resolved.

neilmb, 2015-04-02:
Version 0.4.4 doesn't fix the problem. The new Arduino 1.6.2 IDE appears to move the cores out of /Applications/Arduino.app entirely. I found the Arduino cores in $HOME/Library/Arduino15/packages/arduino/hardware/avr/1.6.2/boards.txt.  It looks like the Arduino IDE is evolving rapidly in this area right now and perhaps a configuration option to specify a directory where to look would be appropriate.

scottdarch, 2015-04-02:
Verified this broke (again) with 1.6.2. 
1.6.1 does work.

scottdarch, 2015-04-02:
Wow, yeah. 1.6.2 is like a total rewrite. Someone needs to tell the Arduino guys to be bumping the major numbers when they break stuff like this.

1.6.1 is working. I'll have to get to 1.6.2 support when I can. It may be a week or two before I have time.

KarlZeilhofer, 2015-04-02:
They have also splitted up the hardware setup into board and crystal/voltage. 
I'm working with an Arduino Pro 3.3V, 8MHz. Here are different Versions available (5V and 3.3V, 8MHz and 16Mhz)
I think the .inorc needs another keyword beside "board-model".

rajanjamu, 2015-07-10:
I still get the same error. Is there anything I am missing. 

Rajans-MacBook-Pro:beep Lemon$ ano build
Searching for Board description file (boards.txt) ... FAILED
Board description file (boards.txt) not found. Searched in following places: 
  - /Applications/Arduino.app/Contents/Java/hardware/**
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/**
  - /usr/local/share/arduino/hardware/**
  - /usr/share/arduino/hardware/**

Gasman2014, 2016-05-06:
Seems to find the boards now but the core libraries have been moved too....

```
[xxxx@johns-mbp] Test $ and build
Searching for Board description file (boards.txt)  in  ['libraries/**/hardware/**', '/Users/johnpateman/Documents/Arduino/**/hardware/**', '/Users/johnpateman/Arduino/**/hardware/**', '/Users/johnpateman/.arduino15/**/hardware/**', '/Users/johnpateman/.arduino/**/hardware/**', '/Applications/Arduino.app/Contents/Java/**/hardware/**', '/Applications/Arduino.app/Contents/Resources/Java/**/hardware/**', '/usr/local/share/arduino/**/hardware/**', '/usr/share/arduino/**/hardware/**']
found multiple:
  - /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/boards.txt
  - /usr/local/share/arduino/hardware/arduino/boards.txt
Searching for Arduino lib version file (version.txt)  in  ['libraries/lib', '/Users/johnpateman/Documents/Arduino/lib', '/Users/johnpateman/Arduino/lib', '/Users/johnpateman/.arduino15/lib', '/Users/johnpateman/.arduino/lib', '/Applications/Arduino.app/Contents/Java/lib', '/Applications/Arduino.app/Contents/Resources/Java/lib', '/usr/local/share/arduino/lib', '/usr/share/arduino/lib']
/Applications/Arduino.app/Contents/Java/lib/version.txt
Detecting Arduino software version ...  1.6.7 (1.6.7)
Searching for Arduino core library  in  ['/usr/local/share/arduino/hardware/arduino/cores/arduino']
FAILED
Arduino core library not found. Searched in following places:
  - /usr/local/share/arduino/hardware/arduino/cores/arduino
```