scottdarch/Arturo #7: compatibility to Arduino IDE

KarlZeilhofer, 2015-04-02:
Hi, 

great that you keep the development of this useful CLI going on!

On my whishlist would be the compatibility to the Arduino IDE folder-structure. 
When I want to open the sketch.ino in the Arduino IDE, it complains about that it is not in the folder named Sketch.
Could this be possible for the future?

greets, Karl

scottdarch, 2015-04-02:
What version of the IDE are you using?
I think Arturo will build if you are using the IDE compatible folder structure. What is it you are looking for from Arturo specifically?

KarlZeilhofer, 2015-04-07:
Hi Scott, 

I am a bit confused. 
I am not really into the details of all the rules for making a proper multi-file arduino-ide compatible project. 
For my current project, the 'WolkenThermometer' (https://github.com/KarlZeilhofer/WolkenThermometer-firmware) I only used ino for writing the program (until now, I haven't tried Arturo). 

Now I am trying to make it compatible to the Arduino IDE. 
I think the first step is to move all the .ino and .h files into the folder /src/sketch/ since the IDE wants to have the main .ino file in a folder with the same name, which is in our case the sketch.ino file. 

Then the problem with the folder structure seams to be fixed, and it still builds with ino without problems. But then in the IDE new problems about include-files appear. E.g. 

    In file included from sketch.ino:24:0:
    helper.h:28:28: fatal error: SoftwareSerial.h: No such file or directory
    #include <SoftwareSerial.h>

What is the suggested way here?
I just want that the project can be compiled with ino (or arturo) and the IDE without any modifications, since I am working with Ino, and most people are working with the IDE. 

Thanks for your help in advance, 
Karl

scottdarch, 2015-04-07:
Let me try your project tonight (I'm PDT/GMT-7) and get back to you.

scottdarch, 2015-04-08:
Okay: I've got two PRs to add this feature for the 1.6.1 and earlier IDE (remember 1.6.2 is still broken for Arturo).

The first PR is for Arturo 0.4.5 to add --source_dir as an optional argument.
The second is back in your KarlZeilhofer/WolkenThermometer-firmware repo to show you fixes you'll need to that project to allow it to compile with the Arduino IDE.

Let me know if this works and let me know when I can buy a thermometer ;-)

KarlZeilhofer, 2015-04-08:
Dear Scott, 

thank you very much for your help. 
I missed, that in the aes256 and sha1 libs the once-statement was missing - thanks for fixing this. 

I think I did all your suggestlions, but ran into some errors. 
First I made the symlinks:
```
karl@karl-ThinkPad-W520:~/programs/arduino-1.6.1/libraries$ ln -s /home/karl/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware/lib/sha1/
karl@karl-ThinkPad-W520:~/programs/arduino-1.6.1/libraries$ ln -s /home/karl/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware/lib/aes256/
karl@karl-ThinkPad-W520:~/programs/arduino-1.6.1/libraries$ ln -s /home/karl/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware/lib/OneWire/
```

I installed Arturo from the develop branch. 
Here the procedure for your reference: 
```
git clone -b develop https://github.com/scottdarch/Arturo.git
cd Arturo
sudo make install
```
In addition I coppied the ~/.inorc to ~/.anorc and changed the dist-path and the board-name from pro328 to pro
```
[build]
board-model = pro
arduino-dist = /home/karl/programs/arduino-1.6.1/
 
[upload]
board-model = pro
arduino-dist = /home/karl/programs/arduino-1.6.1/
serial-port = /dev/ttyUSB0
 
[serial]
serial-port = /dev/ttyUSB0
```
BTW, is the a way to specifiy the crystall and/or cpu?

Then I merged your pullrequest of the WolkenThermometer into the master branch (done on github). 
I pulled the reop to my local machine with 
```
git pull
```

Then I tried your suggested ano invokation:

```
karl@karl-ThinkPad-W520:~/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware$ ano build -s sketch
Searching for Board description file (boards.txt) ... found multiple: 
  - /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/boards.txt
  - /home/karl/programs/arduino-1.6.1/hardware/arduino/sam/boards.txt
Searching for Arduino lib version file (version.txt) ... /home/karl/programs/arduino-1.6.1/lib/version.txt
Detecting Arduino software version ...  1.6.1 (1.6.1)
Searching for Arduino core library ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/cores/arduino
Searching for Arduino variants directory ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/variants
Searching for Arduino core libraries ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/libraries
Searching for Arduino standard libraries ... /home/karl/programs/arduino-1.6.1/libraries
Searching for Arduino user libraries ... FAILED
Arduino user libraries not found. Searched in following places: 
  - libraries/libraries
  - /home/karl/Documents/Arduino/libraries
```

And the compilation with the IDE (1.6.1) doesn't work either...
Please note, that the installed IDE is V1.0 from the Ubuntu Repos (Kubuntu 14.10). I hope, that the IDE 1.6.1 doesn't mix something up here...

scottdarch, 2015-04-08:
Ah, you are on Linux. I believe the user libraries are supposed to be under
~/Arduino/sketchbook/libraries or something like that (see
https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use/installing-a-library-on-linux
for an example). I've installed a VM to test Arturo on Linux. If ano still
doesn't work I'll have another patch for it tonight ;-)

scottdarch, 2015-04-09:
Okay, I just got this to work on Ubuntu 14.04 with Arduino 1.6.1. Arturo develop now has the change to properly look under ~/Arduino for user libraries. For the IDE you simply need to do you ln -s from ~/Arduino/libraries and everybody is happy!

KarlZeilhofer, 2015-04-09:
Dear Scott, 

thank you very much for your effort and support! I would like to send you a WolkenThermometer as a gift, if you come up for the shipping costs :) Please contact me via eMail for the details about that. 

Thanks for fixing the discrepancy between the ano and IDE lib-search-path. Now it compiles in the IDE. 
Now I still ran into the problem with ano and the unspecified MCU and clock-frequency for the board 'pro', which has 4 different versions (see https://github.com/scottdarch/Arturo/issues/5#issuecomment-88860111 )

```
karl@karl-ThinkPad-W520:~/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware$ ano build -s sketch
Searching for Board description file (boards.txt) ... found multiple: 
  - /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/boards.txt                                                                                                                                                                                                         
  - /home/karl/programs/arduino-1.6.1/hardware/arduino/sam/boards.txt                                                                                                                                                                                                         
Searching for Arduino lib version file (version.txt) ... /home/karl/programs/arduino-1.6.1/lib/version.txt
Detecting Arduino software version ...  1.6.1 (1.6.1)
Searching for Arduino core library ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/cores/arduino
Searching for Arduino variants directory ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/variants
Searching for Arduino core libraries ... /home/karl/programs/arduino-1.6.1/hardware/arduino/avr/libraries
Searching for Arduino standard libraries ... /home/karl/programs/arduino-1.6.1/libraries
Searching for Arduino user libraries ... /home/karl/Arduino/libraries
Searching for make ... /usr/bin/make
Searching for avr-gcc ... /home/karl/programs/arduino-1.6.1/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /home/karl/programs/arduino-1.6.1/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /home/karl/programs/arduino-1.6.1/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /home/karl/programs/arduino-1.6.1/hardware/tools/avr/bin/avr-objcopy
Traceback (most recent call last):
  File "/usr/local/bin/ano", line 4, in <module>
    __import__('pkg_resources').run_script('ano==0.4.5', 'ano')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1438, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/EGG-INFO/scripts/ano", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/runner.py", line 79, in main
    args.func(args)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 291, in run
    self.setup_flags(args)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 154, in setup_flags
    mcu = '-mmcu=' + board['build']['mcu']
KeyError: 'mcu'
```

## Here some additional thoughts about IDE and Arturo:
I think, one thing is that I get the compilations running with these and that modifications. 
But because Arduino stands for simplicity, it would be perfect, when it just compiles out of the box. 

ino had no problem to compile the project, when the source files were under /src/ and the libs were under /lib/
It is a very intuitive way of project organisation. 

why does ino or Arturo not use the folder 'sketch' instead of src? then at least the folder structure for the source files would be compatible with the IDE.

And the IDE doesn't support any way for user-libraries, which are just in the project directory?

I think, it would be very great, when a user can just clone the git-repo, and run 'ano build' or open the sketch.ino in the IDE and everthing is working (without the need for symlinks or anything like that). Therefore also a .anorc in the project-dir would be fine, where at least the board and cpu type are specified, and override the user-defaults in ~/.anorc

When the thing with the user-libraries is not supported by the IDE, there is nothing you can do within the Arturo project. But you can definately change the source-code folder from src to sketch. Then one of the two problems would be solved. 

Greets from Austria, 
Karl

scottdarch, 2015-04-10:
Thanks for all this input. It's really helpful. 
I'll enter a bug to fix support for the pro and other boards with variants. That was introduced before 1.6.2 so I can do that before the big rewrite.

As for the src/sketch; I agree with you completely. I want to support IDE/Arturo interoperability without extra arguments and I will enter a new issue to track this. It's a bit slippery though because "sketch" isn't a special word for the IDE. The only reason it wants your .ino file in a folder called sketch is because the main .ino is called sketch (i.e. foo.ino would go under foo/foo.ino). This is the convention the IDE is using to distinguish the main sketch from the other "tabs" it opens for each additional .ino file in the same folder. That said I should be able to write logic that searches for {token}/{token}.ino to figure this out.

scottdarch, 2015-04-10:
Okay: #8 now includes --cpu which will let you build with this command line override. Sorry for the extra arguments but this one may be required when working with the Pro unless picking the first CPU as a default would be acceptable. 

I also opened up #11. Digging through this part of the code for the first time I now realize ino didn't do the makefile generation right. By using the IDE's platform.txt files Arturo should be far less fragile going forward.

Please let me know if this now lets you build. If so I'll merge this PR into master and get started on the 1.6.2 rewrite.

KarlZeilhofer, 2015-04-10:
Dear Scott, 

## new version works
I installed your new develop branch (V0.4.5) and now it compiles without any errors, great work!
Instead of calling ```ano build --cpu 8MHzatmega328``` which works fine, I added the cpu parameter to the ~.anorc

### my .anorc:
```
[build]
board-model = pro
cpu = 8MHzatmega328
arduino-dist = /home/karl/programs/arduino-1.6.1/
 
[upload]
board-model = pro
cpu = 8MHzatmega328
arduino-dist = /home/karl/programs/arduino-1.6.1/
serial-port = /dev/ttyUSB0
 
[serial]
serial-port = /dev/ttyUSB0
```
## avrdude.conf missing
Now one problem still exists: it seams, that the search path for the avrdude.conf is wrong. 
ano complains about that is cannot find it:
```
karl@karl-ThinkPad-W520:~/WifiSensor/firmware/ino-projects/wolkenthermometer-firmware$ ano upload
Searching for avrdude.conf ... FAILED
avrdude.conf not found. Searched in following places: 
  - /home/karl/programs/arduino-1.6.1/hardware/tools
  - /etc/avrdude
```
I found this file in my Arduino disto in ```/home/karl/programs/arduino-1.6.1/hardware/tools/avr/etc/avrdude.conf``` 

## source-dir
I know, that 'sketch' is no keyword. 
A possible way for better compatibility would be your suggestion to search for {token}/{token}.ino
But I think a fixed name here is enough, don't you think so? e.g. src/src.ino or sketch/sketch.ino
As far as I understand the logic of ano, it isn't compatible with IDE projects any way, since it wants to have a separate project-folder, and within that there must be the lib- and src-folder. also the .build_ano goes into that. 

Please keep in mind, that there are 2 ways of compatibility: 
1. open an ano-project in the IDE, and 
2. compile and upload an IDE-project with ano. 

If your goal is compatibility as good as possible both ways should be supported. 
Therfore another folder-structure is needed. Let me show, what I would suggest:
```
.
└── arduino-projects
    └── wolkenthermometer_firmware
        ├── .build_ano
        │   ├── pro-6c1f009f
        │   │   ├── aes256
        │   │   │   ├── aes256.d
        │   │   │   ├── aes256.o
        │   │   │   ├── dependencies.d
        │   │   │   └── libaes256.a
        │   │   ├── arduino
        │   │   │   ├── abi.d
        │   │   │   ├── abi.o
        │   │   │   ├── CDC.d
        │   │   │   ├── CDC.o
        │   │   │   ├── dependencies.d
        │   │   │   ├── HardwareSerial0.d
        │   │   │   ├── HardwareSerial0.o
        │   │   │   ├── HardwareSerial1.d
        │   │   │   ├── HardwareSerial1.o
        │   │   │   ├── HardwareSerial2.d
        │   │   │   ├── HardwareSerial2.o
        │   │   │   ├── HardwareSerial3.d
[... 121 lines cut ...]
        ├── defs.h
        ├── .gitignore
        ├── helper.h
        ├── helper.ino
        ├── idle.h
        ├── idle.ino
        ├── LICENSE
        ├── pins.h
        ├── pins.ino
        ├── README.md
        ├── serialmenu.h
        ├── serialmenu.ino
        ├── settings.h
        ├── sketch.ino
        ├── temperature.h
        ├── temperature.ino
        ├── timer.h
        ├── timer.ino
        ├── voltage.h
        ├── voltage.ino
        ├── wolkenthermometer_firmware.ino
        ├── wt-wifi.h
        └── wt-wifi.ino
```
I must admit, that I have no idea, what the IDE does with the subfolders, which appear then on the same level as the source-files.

scottdarch, 2015-04-11:
Okay, try develop now. I think it should work.

__Note I corrected the source_dir argument to be source-dir__

Also note that you can put a file 'ano.ini' with the same structure as a ~/.anorc under the WolkenThermometer-firmware/ root. You can also add source-dir to it:

(WolkenThermometer-firmware/ano.ini)
```
[build]
board-model = pro
cpu = 8MHzatmega328
arduino-dist = /home/karl/programs/arduino-1.6.1/
source-dir = sketch

[upload]
board-model = pro
cpu = 8MHzatmega328
arduino-dist = /home/karl/programs/arduino-1.6.1/
serial-port = /dev/ttyUSB0

[serial]
serial-port = /dev/ttyUSB0
```

Doing this I was able to build and upload to my Arduino mega from Ubuntu 14 (virt) which has variants. 

Let me know if this clears it all up!

KarlZeilhofer, 2015-04-11:
Dear Scott, 

I've changed the search path again, now it works on my machine!
I've created a pull request. 

greets, 
Karl

scottdarch, 2015-04-11:
Thanks. I'm closing this issue with the caveat that the IDE support stops at 1.6.1. Support for 1.6.2 and newer is ongoing work.

closed 2015-04-11 by scottdarch