scottdarch/Arturo #13: Arduino core library not found

Sman25, 2015-04-17:
I found this repository when I was searching for a fix to a problem with ino.  However, while it fixed my original problem, another one was created.  When I attempt to `ano build` I get this message:

Searching for Arduino core library ... FAILED
Arduino core library not found. Searched in following places: 
  - /usr/local/share/arduino/hardware/arduino/cores/arduino

Is there a problem with my Arduino IDE installation? Or is this a known issue? Thanks in advance for any help.

scottdarch, 2015-04-17:
Arturo currently only works with the 1.6.1 version of the Arduino IDE. Changes introduced in 1.6.2 are requiring a complete rewrite of Arturo. This rewrite is in progress (you can follow along on the [feature/ide_1_6_2 branch](https://github.com/scottdarch/Arturo/tree/feature/ide_1_6_2 branch) but it's only me doing it and it could be awhile (I do have a day job after all ;-)

Closing as a duplicate of #5

closed 2015-04-17 by scottdarch

hachpai, 2015-05-12:
Hello,
I still have the issue, with 1.6.1... Why does it looks there for core??

hachpai@LAlhambra example % ano build                                                                                                                                            [0]
Searching for Board description file (boards.txt) ... /usr/share/arduino/hardware/arduino/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.6.1 (1.6.1)
Searching for Arduino core library ... FAILED
Arduino core library not found. Searched in following places: 
  - /usr/share/arduino/hardware/arduino/cores/arduino

Mausy5043, 2015-05-12:
I have a similar problem. I'm using v1.0.1 of the IDE on a Raspberry Pi and I got:
```
$ ano build
Searching for Board description file (boards.txt) ... /usr/share/arduino/hardware/arduino/boards.txt
Searching for Arduino lib version file (version.txt) ... /usr/share/arduino/lib/version.txt
Detecting Arduino software version ...  1.0.1 (1.0.1)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/cores/arduino
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/variants
Searching for Arduino core libraries ... FAILED
Arduino core libraries not found. Searched in following places:
  - /usr/share/arduino/hardware/arduino/libraries
  - /usr/share/arduino/hardware/arduino/libraries
```

I put a band-aid on that by a symlink to the libraries:
```
$ cd /usr/share/arduino/hardware/arduino/
$ sudo ln -s /usr/share/arduino/libraries/ libraries
```

Now I get another error. That needs to be patched. 

@hachpal : I think you'll could approach this in the same way for now. Find the right location of the directory for your version of the IDE and then put a symlink to that where Arturo is expecting it.

@scottdarch : I first tried patching my problem by changing `environment.py` but it appears not to be that straightforward. If you already know where the paths to be searched are stored I'd appreciate you sharing that knowledge. It would spare me a lot of time reverse-engineering the code.

Mausy5043, 2015-05-12:
Apparently a user library directory is required:

```
Searching for Arduino user libraries ... FAILED
Arduino user libraries not found. Searched in following places:
  - libraries/libraries
  - /home/pi/Documents/Arduino/libraries
  - /home/pi/Arduino/libraries
```

Mausy5043, 2015-05-12:
To get rid of the symlink I've traced the problem to line 136 of ano/commands/build.py .
Kind of stuck from there. It's late. :sleeping:

scottdarch, 2015-05-13:
Sorry all. I'll try to get on this thread tomorrow night. My day job's been busy. 

Checkout our latest release: https://itunes.apple.com/us/app/paper-by-fiftythree/id506003812?mt=8

Mausy5043, 2015-05-13:
No need to apologise.

scottdarch, 2015-05-14:
Awesome all. I found an old ino bug! See issue #29 . Patch will be up shortly.

scottdarch, 2015-05-14:
Please take a look at #30 @hachpai . If this fixed your problem please response in issue #29

23Pstars, 2016-02-08:
This issue still exists, I have clone latest `ano` version (master), using Raspberry Pi
```
Detecting Arduino software version ...  1.0.5 (2:1.0.5+dfsg2-4)
Searching for Arduino core library ... /usr/share/arduino/hardware/arduino/cores/arduino
Searching for Arduino variants directory ... /usr/share/arduino/hardware/arduino/variants
Searching for Arduino core libraries ... FAILED
Searching for Arduino standard libraries ... /usr/share/arduino/libraries
Searching for Arduino user libraries ... FAILED
```

scottdarch, 2016-02-09:
Is there a repository I can pull to try to reproduce this problem?

scottdarch, 2016-02-12:
Oh! Sorry. These failures can be ignored unless you actually had user or
core libraries you needed. I just tried this on my pi and all seems well.

23Pstars, 2016-02-13:
It's strange, I can run the Arduino IDE on X windows but still missing Arduino core in `ano` build.