scottdarch/Arturo #14: Arduino core libraries not found. (1.0.6 and 1.6.1)

sgtsquiggs, 2015-04-22:
```
Arduino core libraries not found. Searched in following places:
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/libraries
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/libraries
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/libraries
```
Libraries are actually at
```
/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/libraries
```

scottdarch, 2015-04-22:
Hmm. What OS are you using? Do you have a github project I can pull and try building myself?

sgtsquiggs, 2015-04-22:
I had to build clean again and it kind of works... instead I get a different error!

With 1.0.6, I was using the nano328 board. It seems with 1.6.1 that is no longer an option?

Using 'nano':

```
Searching for Board description file (boards.txt) ... found multiple:
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/boards.txt
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/boards.txt
Searching for Arduino lib version file (version.txt) ... /Applications/Arduino.app/Contents/Resources/Java/lib/version.txt
Detecting Arduino software version ...  1.6.1 (1.6.1)
Searching for Arduino core library ... /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/cores/arduino
Searching for Arduino variants directory ... /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/variants
Searching for Arduino core libraries ... /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/libraries
Searching for Arduino standard libraries ... /Applications/Arduino.app/Contents/Resources/Java/libraries
Searching for Arduino user libraries ... /Users/matthew/Documents/Arduino/libraries
Searching for make ... /usr/bin/make
Searching for avr-gcc ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-objcopy
Traceback (most recent call last):
  File "/usr/local/bin/ano", line 4, in <module>
    __import__('pkg_resources').run_script('ano==0.4.5', 'ano')
  File "build/bdist.macosx-10.10-x86_64/egg/pkg_resources/__init__.py", line 696, in run_script
  File "build/bdist.macosx-10.10-x86_64/egg/pkg_resources/__init__.py", line 1614, in run_script
  File "/usr/local/lib/python2.7/site-packages/ano-0.4.5-py2.7.egg/EGG-INFO/scripts/ano", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/site-packages/ano-0.4.5-py2.7.egg/ano/runner.py", line 79, in main
    args.func(args)
  File "/usr/local/lib/python2.7/site-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 296, in run
    self.setup_flags(args)
  File "/usr/local/lib/python2.7/site-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 155, in setup_flags
    mcu = '-mmcu=' + BoardModels.getValueForVariant(board, boardVariant, 'build', 'mcu')
  File "/usr/local/lib/python2.7/site-packages/ano-0.4.5-py2.7.egg/ano/environment.py", line 403, in getValueForVariant
    raise KeyError("Are you missing --cpu %s" % (str(boardsDict['menu']['cpu'].keys())))
KeyError: "Are you missing --cpu ['atmega168', 'atmega328']"
```

scottdarch, 2015-04-22:
Ah, the error message is telling you what to do: you need to tell it which variant of the nano you are using. add either
```
--cpu atmega168
```
or
```
--cpu atmega328
```
depending on the type of nano you are building for.

I'm actively working on Arturo 2.0 which will have much nicer error messages than this one. Sorry for the confusion.

closed 2015-04-22 by scottdarch

sgtsquiggs, 2015-04-22:
I can add "cpu = atmega328" in ano.ini too. Perfect.

scottdarch, 2015-04-22:
yep. Happy hacking! Arturo 2.0 is well under way and I hope to have 1.6.2 +
support and many other improvements ready before too long. It's just me
right now so perhaps mid summer?

scottdarch, 2015-04-22:
closed as duplicate of #5

Sman25, 2015-04-23:
I'm very interested in helping develop this project.  I'm not sure how much I'll be able to do, as I am quite busy with school, but I would still like to try and contribute.  If there's anything I can do please let me know.

scottdarch, 2015-04-23:
Thanks! I'll put together a contributing wiki tonight. You can get started by forking the repo and looking at the work I'm doing on the feature/ide_1_6_2 branch. Everything under the Arturo2 package is the rewrite in progress. Also take a peek at https://github.com/scottdarch/Arturo/wiki/Arturo2-Design

scottdarch, 2015-04-24:
Please contact me via email and we can discuss how to get you started. You'll find a mailto: link on my blog (http://32bits.io/)