scottdarch/Arturo #24: Homebrew package

szymonkaliski, 2015-05-08:
I made one for ino, maybe we should create one for arturo as well?

https://github.com/Homebrew/homebrew/commit/d838fe4988c41a952fb9c288c0fdc6fe06e0aa67

scottdarch, 2015-05-08:
Yeah. I was going to do this once Arturo 2 was ready but I have no problem with doing it for Arturo 1.0. It's just a little frustrating since 1.0 is completely broken for the newest Arduino IDEs.

szymonkaliski, 2015-05-09:
Ok, so I think it would be better to wait, I'm not sure if putting something not working with the recent Arduino IDE on brew is the best idea. I think most users (me included) use brew to get something up and running as quickly as possible, and downgrading Arduino IDE doesn't really match this description :)

scottdarch, 2015-05-09:
Yep. I agree completely. I'm actively working on Arturo2 on the develop branch. It should be ready for a beta release by mid to late summer unless I find some collaborators to accelerate the process.

I'll keep this issue open with the 2.0 milestone and ping you when I'm ready.

Thanks for the help.

bfontaine, 2016-02-26:
Is there anything new on this? I came here to ask for a stable version in order to get Arturo in Homebrew :smiley: 

Note we’ll need a *stable* release for Homebrew; beta versions are not accepted in the core.