scottdarch/Arturo #33: Setup Travis CI and PiPy upload

scottdarch, 2015-05-20:
http://docs.travis-ci.com/user/deployment/pypi/

photodude, 2015-05-21:
I mentioned these items elsewhere, but I figure it belongs here so interested developers can discuss this. (This was from a post that requested feedback on pain points from using Arturo1 with Travis CI)

I would say the first pain point is dependencies. I think ideally when installing Arturo with Travis CI Arturo should get the Arduino IDE as a git submodule and should install the other PIP dependencies without the need to specify those dependencies in the .travis.yml file.

The second pain point is the default board selection is currently failing on Travis CI builds, forcing the need to specify the board. 

The good thing about being able to specify the boards is this allows doing a multi-board environment  for testing on Travis CI (like doing tests for various PHP versions, only it's tests against various supported boards)

photodude, 2015-05-29:
@scottdarch Seems there are some issues when trying to test against a custom arduino library and the library example sketch. Error `No project found in this directory` even though there is an existing .ino file in that directory. See [this Travis build](https://travis-ci.org/photodude/DualVNH5019MotorShieldMod3/builds/63823158)

scottdarch, 2015-05-30:
arturo 1.0 still has the odd requirement of a "src" directory. You can get around this using the --source-dir directive. When I tried that I ran into two more issues: first, when building the atmega you need to specify the cpu:

```
ano build --board-model mega --source-dir . --cpu atmega1280
```

The next problem for me was missing libraries but I'm assuming these are installed for your environment. Let me know if this works.

photodude, 2015-05-31:
I assume the library is installed as the build is the library and test against the examples in the library. So with the changes you mentioned I am also getting a `Searching for Arduino user libraries ... FAILED` notice
and `.build_ano/mega/examples/Test_Motors/Test_Motors.cpp:2:40: fatal error: DualVNH5019MotorShieldMod3.h: No such file or directory` from the example's include statement.

scottdarch, 2015-06-02:
Hmm, yeah. I don't think 1.0 was ever designed to build examples properly.
You can open an issue and I'll try to take a look at it this week.

chrisspen, 2015-12-17:
+1 registering on PyPI. I'd love to be able to install via a versioned pip release instead of Github.

scottdarch, 2015-12-17:
I'm reluctant to promote version 1.0 since it's basically deprecated. It doesn't work with the latest Arduino IDEs and is becoming increasingly out of date as a result. If I'm able to find the time and/or attract collaborators to complete v2 we will definitely do this.