scottdarch/Arturo #34: Library examples fail to build

photodude, 2015-06-02:
When the build is a library and tests against the examples in the library, the build fails.

Currently using the --source-dir directive for the build, since arturo 1.0 has the odd requirement of a "src" directory.

The library fails to be detected, build gives a `Searching for Arduino user libraries ... FAILED` notice
and `.build_ano/mega/examples/Test_Motors/Test_Motors.cpp:2:40: fatal error: DualVNH5019MotorShieldMod3.h: No such file or directory` from the example's include statement.
tested against both `<>` and `""` include statements
(issue opened at the request of @scottdarch )

photodude, 2015-06-06:
Possibly related issue from the INO project

https://github.com/amperka/ino/issues/164

scottdarch, 2015-06-06:
No eta on this fix yet. I'm running a little low on bandwidth right now.

photodude, 2017-06-08:
@scottdarch I assume this is still an issue. I found another solution for my basic level testing. As I'm no longer following, shall I close this?