scottdarch/Arturo #38: Ano build builds to weird folder name, then upload can't find files.

Robotto, 2015-06-11:
After a clean build:
$ ls .build_ano/
environment-ano.pickle  leonardo-dd8692bf  uno-dd8692bf

and then after an attempt at upload:
avrdude: can't open input file .build_ano/leonardo/firmware.hex: No such file or directory

scottdarch, 2015-06-11:
huh. Do you have a github repo I can pull to try to reproduce?
Does the uno upload work?
Also, what platform are you on (Windows, Linux, Mac)?

Robotto, 2015-06-13:
The repo is here: https://github.com/418Teapot/pdp/tree/master/Scale_Firmware
I'm on linux.

Will test on an UNO next week.

Robotto, 2015-06-16:
The uno upload fails as well.

scottdarch, 2015-06-17:
Sorry for the delay. My day job has been taking a lot of time this week. I will look at this at some point.

scottdarch, 2015-06-17:
Okay. I did not reproduce this on my mac. I'll have to bring up a 'nix virt to try a reproduce.

Robotto, 2015-06-17:
No need to apologize, you're doing a really good job, and the problem is relatively easily worked around by renaming some folders after each build, eg.:
"ano clean && ano build && mv .build_ano/leonardo-dd8692bf .build_ano/leonardo"

scottdarch, 2015-06-17:
Okay; no repro on Umbuntu. I was able to build and upload to my leonardo. 
1. what version of the arduino IDE are you using?
2. are you just running `ano build` to build the hex file?

Robotto, 2015-06-22:
1. I'm using 1.0.6
2. Yes.

It looks like the distHash (environment.py, line 373) does something weird, but this can be due to the fact that i have defined a custom location for the arduino environment.. have a look at my ~/.anorc: 

´[build]
arduino-dist = /home/robotto/Hack/arduino-1.0.6/

[upload]
serial-port = /dev/ttyACM1

[serial]
serial-port = /dev/ttyACM1
'

Robotto, 2015-06-22:
It may just be an indentation error: https://github.com/scottdarch/Arturo/pull/42

scottdarch, 2015-06-23:
Is it possible that the path/shell you use to build is different then the one you use to upload?

scottdarch, 2015-06-23:
Pull the latest from master and try it. I added https://github.com/scottdarch/Arturo/commit/2d042121167b97d83647d0229d66a1320af549c6

Robotto, 2015-06-23:
Still no deal, but we're getting closer.. 
´´ano build´´ still creates an empty ".build_ano/leonardo" folder and then finds nothing inside when looking for the hex file.

DefProc, 2015-07-08:
Just to confirm equivalent behaviour on Ubuntu using --arduino-dist. 

If I run `ano build -d ~/.bin/arduino-1.6.0 -m uno` the correct folder gets created: `.build_ano/uno` but is empty. 

In addition, the folder `.build_ano/uno-39890d97` is created and the firmware builds there, so `ano upload` fails to find the `firmware.hex`.

Moving the arduino directory to one of the searched paths `/usr/share/arduino/` implements the correct behaviour (building to `.build_ano/uno`)

A quick check for a leonardo build with the same settings shows the same suffix on the build dir: `.build_ano/leonardo-39890d97`

If that helps any.

DefProc, 2015-07-08:
Ah, OK, I have this one figured out for my situation. It's a feature, not a bug! 

Changing the arduino-dist location adds a suffix to the build directory. i.e. `ano build -d ~/.bin/arduino-1.0.2 -m uno` builds to `.build_ano/uno-0b78d22c`
Cause: Operator error
Solution: if using `-d|--arduino-dist`, it must be called on both `ano build` AND `ano upload`. eg.

`ano build -d ~/.bin/arduino-1.6.0 -m uno`
`ano upload -d ~/.bin/arduino-1.6.0`

The suffix is intentionally generated to allow compiled versions for different boards AND for different arduino IDE versions, hence the `.build_ano/<board>[-distHash]` format.

Does that cover your case too @Robotto ?

Robotto, 2015-07-09:
This covers my case exactly. 

Now the issue can be boiled down to: Why doesn't ```ano upload``` read the ```Arduino-dist``` configuration from ~/.anorc

scottdarch, 2015-07-09:
This is good data. I'll see if I can get some time to figure this out now. Unfortunately I'm in NYC next week but perhaps I'll have time on the plane.