scottdarch/Arturo #40: Where to put .ini for defaults?

catskull, 2015-06-22:
I assume that arturo won't look for ino.ini, but ano.ini. Should ano.ini reside in the root directory of the project or somewhere else?

I would be happy to update the readme once I can get a confirmation that this is correct.

Robotto, 2015-06-22:
Yes, the root folder of the project is correct.

scottdarch, 2015-06-23:
Yep. In line 25 of conf.py you'll see three locations where arturo 1.0 looks for the .ini file.
```
'/etc/ano.ini', '~/.anorc', 'ano.ini'
```

catskull, 2015-06-23:
@scottdarch thanks for the confirmation. What is the status with documentation? Is that something that would be valuable for me to contribute?

scottdarch, 2015-06-23:
I'd love to get any help I can, please do contribute. Most of my limited time on this project is spent on the 2.0 port since 1.0 is sort of a dead end. That said, documentation for the 1.0 line will help me maintain these contracts in the 2.0 line.