scottdarch/Arturo #41: Build and upload succeed, but arduino does not change

catskull, 2015-06-22:
I'm trying to build and upload a sketch to my MEGA 2560. Both `ano build` and `ano upload` work correctly, but my arduino does not seem to change. In fact, it appears that it is wiped of any code.

catskull, 2015-06-22:
Build:
```
Searching for Board description file (boards.txt) ... found multiple:
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/boards.txt
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/teensy/boards.txt
Searching for Arduino lib version file (version.txt) ... /Applications/Arduino.app/Contents/Resources/Java/lib/version.txt
Detecting Arduino software version ...  1.0.6 (1.0.6)
Searching for Arduino core library ... /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/cores/arduino
Searching for Arduino variants directory ... /Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/variants
Searching for Arduino core libraries ... FAILED
Searching for Arduino standard libraries ... /Applications/Arduino.app/Contents/Resources/Java/libraries
Searching for Arduino user libraries ... /Users/daviddegraw/Documents/Arduino/libraries
Searching for make ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/make
Searching for avr-gcc ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-gcc
Searching for avr-g++ ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-g++
Searching for avr-ar ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-ar
Searching for avr-objcopy ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-objcopy
Searching for avr-size ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avr-size
src/sketch.ino
Searching for Arduino lib version file (version.txt) ... /Applications/Arduino.app/Contents/Resources/Java/lib/version.txt
Detecting Arduino software version ...  1.0.6 (1.0.6)
Scanning dependencies of src
Scanning dependencies of arduino
Scanning dependencies of SPI
Scanning dependencies of TFTv2
src/TFTv2/font.c
src/TFTv2/TFTv2.cpp
src/sketch.cpp
TFTv2/font.c
TFTv2/TFTv2.cpp
Linking libTFTv2.a
SPI/SPI.cpp
Linking libSPI.a
arduino/avr-libc/malloc.c
arduino/avr-libc/realloc.c
arduino/WInterrupts.c
arduino/wiring.c
arduino/wiring_analog.c
arduino/wiring_digital.c
arduino/wiring_pulse.c
arduino/wiring_shift.c
arduino/CDC.cpp
arduino/HardwareSerial.cpp
arduino/HID.cpp
arduino/IPAddress.cpp
arduino/main.cpp
arduino/new.cpp
arduino/Print.cpp
arduino/Stream.cpp
arduino/Tone.cpp
arduino/USBCore.cpp
arduino/WMath.cpp
arduino/WString.cpp
Linking libarduino.a
Linking firmware.elf
Converting to firmware.hex
Sketch uses 3,062 bytes (1%) of program storage space.
Maxiumum is 258,048 bytes.
Global variables use 705 bytes of dynamic memory.
Maxiumum is unknown.
```

Upload:
```
Searching for stty ... /bin/stty
Searching for avrdude ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/bin/avrdude
Searching for avrdude.conf ... /Applications/Arduino.app/Contents/Resources/Java/hardware/tools/avr/etc/avrdude.conf

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801
avrdude: reading input file ".build_ano/mega2560/firmware.hex"
avrdude: writing flash (3062 bytes):

Writing | ################################################## | 100% 0.44s

avrdude: 3062 bytes of flash written
avrdude: verifying flash memory against .build_ano/mega2560/firmware.hex:
avrdude: load data flash data from input file .build_ano/mega2560/firmware.hex:
avrdude: input file .build_ano/mega2560/firmware.hex contains 3062 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.35s

avrdude: verifying ...
avrdude: 3062 bytes of flash verified

avrdude: safemode: Fuses OK

avrdude done.  Thank you.
```

catskull, 2015-06-22:
It looks like searching for Arduino core libraries fails, yet the build still succeeds. Could that be the culprit?

Robotto, 2015-06-22:
Try it with "ano clean" first, to remove any residual build files.
And could you post a listing of the contents of the .build_ano/ folder after "ano build"?

scottdarch, 2015-06-23:
The failure to find core libraries is not necessarily fatal if your app doesn't rely on them. Arturo 1.0 does support the newer 1.6.1 IDE. You might have better luck with it.

catskull, 2015-06-24:
@Robotto `ano clean

scottdarch, 2015-06-25:
Hmm. There has been one change to ano that could affect compilation:

https://github.com/scottdarch/Arturo/commit/3242b1c4a7e06b66129fb253b59ae5f6b2fef081

Is the Mega a different architecture?

catskull, 2015-06-25:
Do you mean is the mega2560 a different architecture than AVR? The ATmega2560 is AVR architecture.

For you reference: http://www.atmel.com/devices/atmega2560.aspx

s-hell, 2015-07-10:
I have the same problem with arduino uno.

eshansingh, 2016-06-15:
It's 2016 now! It looks like even Arturo has been abandoned.