scottdarch/Arturo #43: Board description file (boards.txt) not found

rajanjamu, 2015-07-11:
I tried everything possible. Went through all the forums and issues on this repository. Seems like this issue has been solved, but not for me. Please help. Kind of frustrated with Arduino IDE.

Searching for Board description file (boards.txt) ... FAILED
Board description file (boards.txt) not found. Searched in following places: 
  - /Applications/Arduino.app/Contents/Java/hardware/**
  - /Applications/Arduino.app/Contents/Resources/Java/hardware/**
  - /usr/local/share/arduino/hardware/**
  - /usr/share/arduino/hardware/**

scottdarch, 2015-07-11:
what version of the Arduino IDE are you using and on what platform?

rajanjamu, 2015-07-12:
Hey @scottdarch, I am using Arduino 1.6.5 on Yosemite OS X (10.10.1).

petrkotek, 2015-11-23:
`README.md` mentions that Arturo supports only "Arduino 1.6.1 or earlier":

> A command-line tool for Arduino 1.6.1 or earlier

scottdarch, 2015-11-24:
Yeah. Sorry. That's where Arturo leaves off for now: 1.6.1. I have been picking away at a v2 but so far I haven't found any collaborators so no telling if I'll have a chance to finish it or not.

Robotto, 2015-11-24:
What do you need?

~Mark

scottdarch, 2015-11-24:
Just someone with Python and/or GNU Make chops to collaborate with. The V2 is about 25% done and with my schedule through next year I'm not going to have much time for it until February or so. At this pace I'd be lucky to have v2 done in time to support Arduino Mille (featuring Atmel's latest 8 qbit processor).