scottdarch/Arturo #50: avrdude.conf location not found on Ubuntu14.04

tomjuggler, 2015-11-17:
Hi, when uploading I received the following error:

Searching for avrdude.conf ... FAILED
avrdude.conf not found. Searched in following places: 
  - /usr/local/share/arduino/hardware/tools/avr/etc
  - /usr/share/arduino/hardware/tools/avr/etc
  - /etc/avrdude

I eventually found my avrdude.conf file at /etc/avrdude.conf instead of in /etc/avrdude folder 
I have made a copy of avrdude.conf and moved it into the /etc/avrdude folder which I had to create first as it did not exist on my system. Once that was done everything works great. 

It is possible that this is an issue unique to my system, which is usually a bit of a mess. I don't remember moving that file however...

scottdarch, 2015-11-18:
Hmm. Not sure. What version of the Arduino IDE are you using?

tomjuggler, 2015-11-19:
Using 1.0.6 
In the notes somewhere I read that Arduino will use the version of avrdude installed on your system. Not sure how or when avrdude was installed by me though, perhaps it would be best to check this on a clean ubuntu system mine is not the best example.

krsmith28p, 2016-01-02:
I had same issue on Wheezy version of Raspbian  - indeed your post allowed me to continue.
I installed Arduino from the repositories, so version 1.01 I believe. 
Using find located it in two places:-
pi@rasp / $ sudo find -name "avrdude.conf"
./usr/share/arduino/hardware/tools/avrdude.conf
./etc/avrdude.conf 

both are one layer up from the folders searched by the build script

scottdarch, 2016-01-06:
Fixed by #54

closed 2016-01-06 by scottdarch