scottdarch/Arturo #51: Ano serial displays garbage

chrisspen, 2015-11-30:
Running `ano serial` shows mangled output like:

    �E��Ģ��բբ�"Ģ�Ģ�"բ�.���.�U�L�'��E�E-E,\�Ӷ���&
�K�\��L��\��፶|��Dr�|��g���|��L��L��Ò����Ub�Ub�Ub�\��\��Eb���G�

whereas running `ino serial` in the same directory shows the correct output like:

    test:499
    test:1000
    test:1500
    test:2000
    test:2500
    test:3001
    test:3502

This is my test program running on an Arduino UNO:

    void setup(){
        Serial.begin(57600); // must match ino.ini
    }

    void loop(){
        Serial.println(String("test:")+String(millis()));
        delay(500);
    }

and this is my ino.ini:

    [build]
    board-model = uno

    [upload]
    board-model = uno
    serial-port = /dev/ttyACM0

    [serial]
    serial-port = /dev/ttyACM0
    baud-rate = 57600

Is this a bug, or is there some additional configuration Ano needs in order to create a serial connection?

Robotto, 2015-11-30:
Try renaming ino.ini to ano.ini

~Mark

chrisspen, 2015-12-01:
Yep, that was it. I didn't see that mentioned in the docs, but I should have thought of it. Thanks.

closed 2015-12-01 by chrisspen