scottdarch/Arturo #52: Syntax error: "(" unexpected

chrisspen, 2015-12-17:
Ano works perfectly on my 64-bit Linux system, but when I installed it and the 32-bit Arduino IDE on a Raspberry PI, `ano build` fails with the error:

    /usr/share/arduino/hardware/tools/avr/bin/avr-gcc: 1: /usr/share/arduino/hardware/tools/avr/bin/avr-gcc: Syntax error: "(" unexpected
    make: *** [.build_ano/leonardo/src/main.d] Error 2
    Make failed with code 2

My main.ino file is simply:

    void setup(){
    }

    void loop(){
    }

yet the file `.build_ano/leonardo/src/main.d` is blank.

In each case, I'm targeting the same Arduino Leonardo device.

How do I configure ano to build on a 32-bit system?

scottdarch, 2015-12-17:
hmm. My Pi is at home. I can try this tonight. What version of the Arduino IDE are you using?

chrisspen, 2015-12-17:
Nevermind, this isn't an Arturo issue. I just realized that the pre-compiled 32-bit Arduino IDE I installed is compiled for 32-bit x86, not ARM. The error message is unintuitive, but it means it's on the wrong architecture. I fixed it by removing the Arduino IDE and installing the older arduino package from the repo, which is compiled for the right architecture.

closed 2015-12-17 by chrisspen