scottdarch/Arturo #53: Build Error

dbhaig, 2015-12-28:
Thank you, in advance, for your assistance.

I have been using ino tools on Ubuntu 12.04 and thought that I would give Arturo a try. 
I create a project directory:
   cd'd into the directory
   ran ano init 
   then ran ano build to check that everything was working

 I get the following error:
...
Converting to firmware.hex
Traceback (most recent call last):
  File "/usr/local/bin/ano", line 5, in <module>
    pkg_resources.run_script('ano==0.4.5', 'ano')
  File "/usr/local/lib/python2.7/dist-packages/setuptools-1.4.1-py2.7.egg/pkg_resources.py", line 492, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/local/lib/python2.7/dist-packages/setuptools-1.4.1-py2.7.egg/pkg_resources.py", line 1350, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/EGG-INFO/scripts/ano", line 6, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/runner.py", line 79, in main
    args.func(args)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 370, in run
    self.check_memory(args)
  File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 290, in check_memory
    data_size = int(re.search('\.data\s+(\d+)', output).group(1))
AttributeError: 'NoneType' object has no attribute 'group'

krsmith28p, 2016-01-02:
I had the same issue. I solved it, at least for the bleep example, by changing  build.py
File "/usr/local/lib/python2.7/dist-packages/ano-0.4.5-py2.7.egg/ano/commands/build.py", line 290 ish,

#if statement added by ks
if (re.search('\.data\s+(\d+)', output))==None:
          data_size=0
else: #the else statement is the original statement as in the original 'build.py' file 
          data_size = int(re.search('\.data\s+(\d+)', output).group(1))
I could not work out exactly what this statement is doing, or where it is looking (to determine the data_size), so why it doesn't find any result I am not sure. The two lines before and after are identical in structure, and seem to work, without any crude hacking by me, so it is probably not an operating system issue.

Mine was using the wheezy version of Rasbian, if that is relevant. 
Explanation and/or a more appropriate fix would be welcomed! I plan to use this on several different headless Pi-s.
KS

scottdarch, 2016-01-04:
Sorry all. I've been out on holiday. I'll try to take a look at this today.

scottdarch, 2016-01-05:
Ah I see it now. This was introduced by a recent change to support the code size prompt after a build. I'll look into your fix @krsmith28p

PR #54 (2016-01-06): Fixing an exception some Ubuntu users were seeing.
Please take a look at this @mvgrimes . I just added some error checking to your regex logic.

Fixes #53
closed 2016-01-06 by scottdarch