scottdarch/Arturo #57: Unrecognized argument in option '-mmcu=cortex-m3' : using wrong makefile

Swapnull, 2016-02-08:
Hi,

I have decided to switch over to an Arduino Due for one of my projects and while trying to run `ano build`, I am getting the error
`avr-g++: error: unrecognized argument in option '-mmcu=cortex-m3'`
I have had a rummage around the files and I believe that the issue may be stemming from Arturo using the makefiles from
`/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/avr/boards.txt` 
instead of those from `/Applications/Arduino.app/Contents/Resources/Java/hardware/arduino/sam/boards.txt`

I am using arduino 1.6.1 and my ano.ini file is as follows

```
[build]
board-model = arduino_due_x_dbg
cpu = atmega16u2

[upload]
board-model = arduino_due_x_dbg
cpu = atmega16u2
serial-port = /dev/cu.usbmodem1421

[serial]
serial-port = /dev/cu.usbmodem1421
```
I have had a look into the Arturo files to see if I can figure this out but it is way out of my league. I assume its a case of putting a clause it to check to see if we are using a Due or not. 

Thanks for any time and effort in advance.

scottdarch, 2016-02-09:
Hmm. Should be easy to add yet another config option ([sam|avr|intel|etc]). Do you have a repo I can pull to reproduce the error and I'll see if I can hack in a fix?

Swapnull, 2016-02-09:
I have given you contributor access to the project. The Arturo project can be found at https://github.com/Swapnull/Olympus/tree/master/zeus.
Thanks

scottdarch, 2016-02-09:
Huh. I found a hardcoded "AVR" that was using the wrong toolchain but when I fix this I still get the error. I'll dig deeper tonight. Thanks for the repo access.

scottdarch, 2016-02-10:
Unfortunately the problems turned out to be more fundamental. The problem with Arturo v1 is that is forms its own command strings instead of using the ones from the Arduino SDK. because of this, even if you do find the right compiler, none of the other switches are correct.

Sorry to be the bearer of bad news. Arturo 2 would solve this but I still don't have a firm date for when I'll get around to finishing it.

You'll have to just switch to using the Arduino commandline if you can handle running Java in your environment:

https://github.com/arduino/Arduino/blob/ide-1.5.x/build/shared/manpage.adoc

closed 2016-02-10 by scottdarch

Swapnull, 2016-02-10:
Ah I see, thanks for looking anyway, I will try find some other way to
compile until Arturo 2 :)