scottdarch/Arturo #58: Does this project support python 3?

fake-name, 2016-02-11:
I've read through all the files in the repository root, and I can't for the life of me determine if this project supports python 3.

It's not indicated in `setup.py`, it's not noted in `README.md` or `INSTALL`, and it's not stated on the github pages webpage.

Short of actually installing it, is there anywhere that actually can tell me if the project is py3k compatible?

fake-name, 2016-02-11:
Aaaaand having installed it, it installs fine on py3k, and then doesn't work.

Please add a minimal version check to the setup:

https://stackoverflow.com/questions/13924931/setup-py-restrict-the-allowable-version-of-the-python-interpreter

PeterMitrano, 2016-03-06:
If you have a link to how to do it, you should make a pull request! Open source only works if people contribute when they want something.