scottdarch/Arturo #63: Upload prints to standard output when everything is OK

AlexandreRio, 2016-04-25:
Hi folks,

I'm currently wrapping ano in a program to automate some unit testing on a real board. It works fine but I noticed that `ano upload` (and the call to avrdude) prints to the error output, which is sad when everything goes well.

Can it be fixed by Arturo or is the problem somewhere else?

```
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f
avrdude: reading input file ".build_ano/uno/firmware.hex"
avrdude: writing flash (7182 bytes):

Writing | ################################################## | 100% 1.17s

avrdude: 7182 bytes of flash written
avrdude: verifying flash memory against .build_ano/uno/firmware.hex:
avrdude: load data flash data from input file .build_ano/uno/firmware.hex:
avrdude: input file .build_ano/uno/firmware.hex contains 7182 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.93s

avrdude: verifying ...
avrdude: 7182 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

closed 2016-05-30 by AlexandreRio

scottdarch, 2016-05-31:
Yeah, sorry. This is avrdude not Arturo.

AlexandreRio, 2016-05-31:
I know, but I actually fixed it in #64 by adding a new option -q on the upload command. :smile: