scottdarch/Arturo #68: include 3rd party library (freetype)

aars, 2016-06-30:
I'm having trouble compiling [Adafruit's GFX Library](https://github.com/adafruit/Adafruit-GFX-Library) through ino and ano, it can not find freetype libs on my Mac OSX system. Freetype is installed through brew.

Arduino IDE _is_ able to compile the sketch correctly, it is able to locate/include/link the lib.

I've tried a lot of different `--cflags` or `--cxxflags` or `--ldflags`, but the `build` always fails with:
```
lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:24:10: error: #include expects "FILENAME" or <FILENAME>
 #include FT_GLYPH_H
```

I can somehow get past this error by running `build --cxxflags="-I/usr/local/include/freetype2 -lfreetype -MM" --cflags="-I/usr/local/include/freetype2 -lfreetype" --ldflags="-I/usr/local/include/freetype2"` _again_, but then I end up with:
```
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld:.build_ano/uno/src/sketch.ino.cpp.o: file format not recognized; treating as linker script
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld:.build_ano/uno/src/sketch.ino.cpp.o:1: syntax error
```
(with Arduino IDE's avr, and with 4.9 avr tools installed through brew).

I can also get past all compiler errors by copying the freetype header files into the `lib` dir of the project dir, but then I end up with linker errors:
```
build_ano/uno/Adafruit-GFX-Library/libAdafruit-GFX-Library.a(fontconvert.c.o): In function `main':
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:115: undefined reference to `FT_Init_FreeType'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:119: undefined reference to `FT_New_Face'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:121: undefined reference to `FT_Done_FreeType'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:126: undefined reference to `FT_Set_Char_Size'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:140: undefined reference to `FT_Load_Char'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:146: undefined reference to `FT_Render_Glyph'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:153: undefined reference to `FT_Get_Glyph'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:194: undefined reference to `FT_Done_Glyph'
/Applications/Arduino.app/Contents/Java/libraries/Adafruit-GFX-Library/fontconvert/fontconvert.c:232: undefined reference to `FT_Done_FreeType'
collect2: error: ld returned 1 exit status
```

How can I configure ino/ano/avr to include and link this (or all) 3rd party libraries from my system?

Thanks.

aars, 2016-06-30:
I think this should be correct:

`pkg-config --libs freetype2` -> `-L/usr/local/Cellar/freetype/2.6.3/lib -lfreetype`
` pkg-config --cflags freetype2` -> `-I/usr/local/Cellar/freetype/2.6.3/include/freetype2`

`ano clean && ano build --cflags="-I/usr/local/Cellar/freetype/2.6.3/include/freetype2" --ldflags="-L/usr/local/Cellar/freetype/2.6.3/lib -lfreetype" -v`

Gives me the linker errors again:
```
.build_ano/uno/Adafruit-GFX-Library/libAdafruit-GFX-Library.a(fontconvert.c.o): In function `main':
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:115: undefined reference to `FT_Init_FreeType'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:119: undefined reference to `FT_New_Face'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:121: undefined reference to `FT_Done_FreeType'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:126: undefined reference to `FT_Set_Char_Size'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:140: undefined reference to `FT_Load_Char'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:146: undefined reference to `FT_Render_Glyph'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:153: undefined reference to `FT_Get_Glyph'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:194: undefined reference to `FT_Done_Glyph'
/Users/aaronh/src/arduino/waterer-ano/lib/Adafruit-GFX-Library/fontconvert/fontconvert.c:232: undefined reference to `FT_Done_FreeType'
```

Any help would be appreciated.

aars, 2016-06-30:
Ah. I can simply remove the whole `fontconvert` dir and it compiles nicely. I'm not sure if the lib will need it. Should I exclude that when compiling?

Any clue as to why the Arduino IDE doesn't stumble over this?

scottdarch, 2016-07-05:
INO unfortunately doesn't use the Arduino compiler directives so it simply didn't inherit this from the configuration as it should have.

Really INO/Arturo 1.0 should be considered EOL. I've moved on from it myself.

aars, 2016-07-07:
@scottdarch What do you use now?

scottdarch, 2016-07-07:
I've just been writing my own makefiles. Arduino does have a commandline build now for what it's worth. There's also http://platformio.org/ but I haven't used this.

closed 2016-07-11 by aars