scottdarch/Arturo #73: Support for splitting in numtiple .ino files

pavelmc, 2016-09-19:
Hi, the OEM Arduino IDE has a feature in which you can split your code in multiple .ino files to segment your code into features in multiple individual files.

Can we add this feature in Arturo?

Looking here https://www.arduino.cc/en/Hacking/BuildProcess I found that the OEM software just concatenate the other .ino files at the end of the main .ino file (sketch.ino in our case) so it must be easy to fix.

Just detect all the .ino files in the src folder and add then to the main sketch.ino as the first step in the compilation.

Cheers.

kd0skh, 2016-10-18:
This might sound kind of dumb, but have you tried using:

`#include inoA.ino`
`#include inoB.ino`

in your main sketch file?

pavelmc, 2016-10-18:
Right, I will try it...
Tnx

pavelmc, 2016-10-19:
No, it does not work.

I have split the section on my code relating to EEPROM handling (just about 4 procedures) and tested the include trick naming it eeprom.ino:
- The correct way is: #include "otherfile.ino" (note the " around the filename)
- Variables declared globally on the main .ino file are not found on the eeprom.ino (the include is placed below the variable declaration) keep in mind that I dont want to include a library I just want to split the code in sections related to functions (eeprom, shields specific, LCD, serial) to gain in organization of the code just like the OEM IDE allows.
- Principal library #includes on the main file are not passed to the eeprom.ino, you have to move the particular ones to the eeprom.ino and it works up to some point; but for others it keep saying "(...) was not declared in this scope"
- Finally I toke a working arduino ide project with split files and moved it to a arturo project keeping the files as it's just renaming the main .ino file to sketch.ino: it does not work, even after #includes of the related files, keep saying the error in the above point.

So basically it don't work, for a new project or for an imported project from the arduino ide tool.

I think it's important to keep compatibility for this feature from the OEM software, and nevertheless it's a useful thing to keep code organized.

I can be a light patching to the code, it's just detect other .ino files on the src folder and concatenate them below he main sketch.ino file BEFORE prep & compilation...

Unfortunately I'm not fluent enough in python or I will get along with a patch for this by this time...

Cheers, Pavel.